AeolusUX/ArkOS-R3XS #1: no drastic? keeps returning to rom screen, not showing under emulators

akidone, 2024-02-10:
as title says, when trying to boot up a NDS rom, it shows the ArkOS loading screen twice, then returns to the selection screen.

Also, it does not show up when i go into the menu where is select my favorite emulator for a device. It shows Nintendo DS, but it just says AUTO.

When checking for the installed cores, it does not show up, either.
I don´t think that this is expected behavior, or am i doing something wrong here?

AeolusUX, 2024-02-11:
Please reference the link from the official wiki. 
https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information

Drastic "is" the default emulator for the NintendoDS on ArkOS and no other emulator. So technically there is just auto (drastic) If you can't run your games it's either a rom issue or bios. Accessing the menu by clicking the L3 button should also show you to the Drastic Menu.

closed 2024-02-11 by AeolusUX

akidone, 2024-02-12:
Thanks @AeolusUX - I guess i had been a bit too tired when doing the initial install. After unzipping the files and placing the actual nds files, everything works as expected... zZzZz