AeolusUX/ArkOS-R3XS #2: R36S - Quick Save and Quit not working

This-will-be-your-username, 2024-02-11:
Not sure if this belongs here or the parent project (https://github.com/christianhaitian/arkos), however, after flashing v2.0 (01/27/2024) and subsequent updates, the save/exit button combo (press right joystick and power) no longer works.

nicmare, 2024-02-12:
> Not sure if this belongs here or the parent project (https://github.com/christianhaitian/arkos), however, after flashing v2.0 (01/27/2024) and subsequent updates, the save/exit button combo (press right joystick and power) no longer works.

Can not confirm this. Works for me with this built. But you have to enable it again in settings.

AeolusUX, 2024-02-12:
On this image it is moved to FN+Power.
Global hotkeys for this build could be found here:
https://github.com/AeolusUX/ArkOS-R3XS/wiki/Global-event-keys-and-emulator-event-keys.

closed 2024-02-12 by AeolusUX