AeolusUX/ArkOS-R3XS #3: Fn button not workingg anymore.

VEMD, 2024-02-13:
After updating to 02/06/2024 verson, the fn button doesn't work anymore. There is a way to enable it or is it a bug?

kvnol, 2024-02-14:
@VEMD The FN button still works, however, Retroarch reverts to the default settings; it needs to be remapped in Retroarch under `Settings > Input > Hotkeys` again.

AeolusUX, 2024-02-14:
> After updating to 02/06/2024 verson, the fn button doesn't work anymore. There is a way to enable it or is it a bug? In retroarch, for example, the only hotkeys working are the global ones.

Yea it seems to be a problem on the 02062024 image I uploaded last. I've fixed this and if you redownload and reflash should be working now.

closed 2024-02-14 by AeolusUX

spamsool, 2024-02-15:
It works now, however when I use FN+Up/Down to change the brightness it also scrolls the Systems in main menu at the same time. Since there is FN+VolUp/Down shortcut to change the brightness maybe the former is not really necessary?

r0ckarong, 2024-02-16:
Ran into another effect yesterday. FN+Down switches the save state slot to a lower one but FN+Up only increases the brightness. I ended up on Slot 963 and had to switch that back from the menu.