AeolusUX/ArkOS-R3XS #9: Black Screen Lights up on boot after mounting new image on sd and keep stuck there

franciscohcf, 2024-02-29:
Tried all morning to get the ArkOS-R3XS to work but i can never boot up, tried multiple sd's (The one that came with the device and a Sandisk one) both gets me to the same result. Also tried every tool i know to do the same process. And trying to go back to other versions of ark are also getting me to never be able to boot.

[image]
That's the state, the screen is on, and keep stuck that way.

r0ckarong, 2024-02-29:
I had a similar thing once. Could you try pressing the reset button (small one below power) when it shows the black screen?

franciscohcf, 2024-02-29:
yeah i did, tried every combination of buttons also to see if something happens, but the screen remains black and flickers from time to time

AeolusUX, 2024-02-29:
It's not because of my image as you stated even the stock arkos image and old versions doesn't work either. So I'm afraid I couldn't help you posting here. This is not what git issues is for.

My guess is it's your sd card, even sandisk cards have problems too and there are too many fake sandisk cards so I wouldn't trust that particular brand. Also you shouldn't have used the stock cards either. 

Try removing all sd cards except the OS card for first boot. If it still doesn't work it's probably your card or your device or how you flash your sd card.

I would also suggest you ask on discord, reddit or any other community regarding this.

franciscohcf, 2024-02-29:
Ok, thanks for the reply.