AeolusUX/ArkOS-R3XS #14: N64 Mupen64Plus Saves + States + Screenshots not working

DeusAbsconditus, 2024-04-04:
Hello,

there seems to be an issue with the N64 emulator Mupen64plus when creating saves, save states, and screenshots.
Upon closer examination, it appears that incorrect paths are specified in the emulator's configuration file (/home/ark/.config/mupen64plus/mupen64plus.cfg).
The entries ScreenshotPath, SaveStatePath, and SaveSRAMPath point to a directory named "roms2".
However, this directory seems to lead nowhere in an ArkOS installation with only one SD card.

It might be necessary to change the entries in the configuration file to point to the "roms" directory instead. This directory should definitely exist.

AeolusUX, 2024-04-05:
Have you tried doing Switch to Main SD? That script changes the mupen64plus.cfg paths back to /roms/ instead of /roms2/

DeusAbsconditus, 2024-04-05:
Where do I find this?

AeolusUX, 2024-04-06:
It's only for this image since I might've tried the switch to sd2 script since that's not working for my 3/30/2024 update, I'll make sure future images are set to default.

closed 2024-04-06 by AeolusUX