AeolusUX/ArkOS-R3XS #19: WiFi not working anymore

Makuchito, 2024-04-17:
After updating to the 4/11/2024 built via the R36S  OTG using a usb-c to USB adapter  (TP Link AC600) WiFi seems to not work anymore.  I read the change log and saw the Kernal files for the wireless folder were updated. Hope this helps others updating from 4/02/2024 AeolusUX. Solution was to wipe SD card and load the previous version.

AeolusUX, 2024-04-17:
Try this

1. Take the attached compressed file ([arkosbackup.tar.gz](https://raw.githubusercontent.com/AeolusUX/ArkOS-R3XS-Updater/main/04112024/wifi/arkosbackup.tar.gz)) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the device.

4. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Tools, hit A, then scroll all the way down to kernel_mod_update., hit A.

5. The unit will reboot again.

6. Starting testing your wifi adapter again as you did here to see if there's any difference.

Makuchito, 2024-04-17:
Thank you! I will try this and update with the results. 👍

Makuchito, 2024-04-17:
Is there a way to extract the wireless folder from the previous build 4_02_2024 and create an arkosbackup.tar.gz to transfer to the SD?

Srafington, 2024-04-17:
Does work for me either, WiFi status remains "Off" when known good dongles are attached.

AeolusUX, 2024-04-18:
> Is there a way to extract the wireless folder from the previous build 4_02_2024 and create an arkosbackup.tar.gz to transfer to the SD?

That's exactly what we're doing here :D Try to do it again I just changed the files contained on the arkosbackup.tar.gz

Srafington, 2024-04-18:
The updated patch works for me

Makuchito, 2024-04-18:
Thanks again @AeolusUX  The backup file worked! WiFi back up and running!

closed 2024-04-18 by Makuchito

barney-tech, 2024-04-19:
Thanks @AeolusUX all working great again. Super appreciated.

BigZoz, 2024-04-26:
I have the same issue after update, i use the following commands
sudo apt-get clean
sudo apt-get update
sudo apt-get upgrade

then Install the Kernel Header File

sudo apt-get install linux-headers-$(uname -r)

problem solved

Also try this

sudo apt install -y realtek-rtl8188eus-dkms

SandGuyTome, 2024-09-05:
Kernel Not Working