AeolusUX/ArkOS-R3XS #20: Games play with no sound when battery is below 20%

cobaltgit, 2024-04-19:
I'm not sure if this is intentional or not and this might need to be passed on to the main ArkOS repo but when I launch a game, the text-to-speech voice plays "Urgent warning: your battery level is atr _n_%" and then the game proceeds to launch albeit with no sound. When the console (R36S in my case) is plugged in prior to launching the game, the sound plays as does normally.

AeolusUX, 2024-04-20:
I think it should be for the main repo there are quite a few reports of this already on the discord. This is because of the battery warning, consider turning it off for now.

cobaltgit, 2024-04-23:
Closing due to identical reports on Discord and re-opening on main repo

closed 2024-04-23 by cobaltgit