AeolusUX/ArkOS-R3XS #22: Can't mount 2nd SDCARD

rabin-io, 2024-05-02:
On insert, I see this message 

```
[ 1222.539861] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 1222.903331] net_ratelimit: 22 callbacks suppressed
[ 1222.903361] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1222.967337] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.031340] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.095337] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.159337] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.223338] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.287339] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.293661] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1223.299553] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1223.347146] mmc1: error -110 whilst initialising SD card
[ 1223.431342] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.495343] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.559340] cdc_ether 1-1:2.0 eth0: kevent 11 may have been dropped
[ 1223.855616] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1223.861485] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1223.883922] mmc_host mmc1: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[ 1224.411636] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1224.417512] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1224.439887] mmc_host mmc1: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[ 1224.967649] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1224.973523] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[ 1224.995910] mmc_host mmc1: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[ 1225.535675] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[ 1225.541572] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
```

AeolusUX, 2024-05-04:
So is it in the compatibility list?

rabin-io, 2024-05-04:
The original I got with the device which is unbranded card, works with the 2nd SDCARD port, but the Sandisk Extream Pro 128G I have is not working on the 2nd port, but it does on the first port.

With the unbranded device, I get this message from the log,

```
[  200.890606] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[  200.972239] mmc_host mmc1: Bus speed (slot 0) = 150000000Hz (slot req 150000000Hz, actual 150000000HZ div = 0)
[  201.314506] dwmmc_rockchip ff380000.dwmmc: Successfully tuned phase to 169
[  201.314550] mmc1: new ultra high speed SDR104 SDXC card at address 0001
[  201.315669] mmcblk1: mmc1:0001 SD 58.2 GiB 
[  201.319499]  mmcblk1: p1
```

AeolusUX, 2024-05-04:
Yeah your error log is not helping, This is not a compatible device for TF2.

closed 2024-05-04 by AeolusUX