AeolusUX/ArkOS-R3XS #31: PortMaster not working

elmoiv, 2024-05-26:
PortMaster and installed ports not working after installing 24/5/2024 update.

elmoiv, 2024-05-26:
Getting:
```
Failed to create secure directory (/run/user/1002//pulse): No such file or directory
```
In all ports installed.

fabinhojcr, 2024-05-27:
Thememaster also broke.

AeolusUX, 2024-05-27:
I've already figured out the issue and will release a fix in a few.

AeolusUX, 2024-05-27:
Try this

Take the attached compressed file [arkosbackup.tar.gz](https://github.com/AeolusUX/ArkOS-R3XS/files/15450467/arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

Once completed, reboot the device.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Tools, hit A, then scroll all the way down to ports-fix, hit A.

The unit will reboot again.

Try your ports again if it works.

fabinhojcr, 2024-05-27:
This solution solved the problem.

closed 2024-05-27 by AeolusUX

chronoss09, 2024-06-14:
> Try this
>
> Take the attached compressed file [arkosbackup.tar.gz](https://github.com/AeolusUX/ArkOS-R3XS/files/15450467/arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress [...]

it work for me too, thanks a lot for the fixe !!!