AeolusUX/ArkOS-R3XS #37: After the update, USB tethering was lost.

jurandy007, 2024-06-11:
I noticed that in this latest version it only charges the phone and does not enable the option to activate USB tethering like the stock system.

Are there any solutions or options in ArkOS for this?

AeolusUX, 2024-06-11:
Nothing’s changed it’s not ArkOS fault. Check your device or cable. Usb Tethering is not advertised to work some do and some don’t nothing we can do about it. It’s best to buy a compatible dongle.

closed 2024-06-11 by AeolusUX

jurandy007, 2024-06-14:
Sorry if I seemed offensive by just asking.

I wanted to say that in the ArkOs System that came with it it worked and in this one it didn't. (same cell phone and cable, yes I tested with different cables)

but thanks anyway

AeolusUX, 2024-06-14:
That’s impossible as nothing has changed on the kernel since then. You might be missing a step on your procedure as the UI has changed other than that i’m reiterating that the best thing to do is get a compatible dongle.

jurandy007, 2024-06-14:
I understand, I already have a Wifi dongle and use it, but I was curious about this function. I even thought it could be something that I didn't activate in the system settings.