AeolusUX/ArkOS-R3XS #39: [Question]Palm game for noob

Fr-Dae, 2024-06-17:
Good morning,
I've probably done something wrong, but I can't run the downloaded games for palm os.

I have checked that I have the latest core for palmOS(mu).
Nothing is displayed in the main menu with the other systems.

I also tried in "retroarch" mode by manually selecting the zip of the game and the core, nothing launches.

If you have any ideas, I'm interested.

AeolusUX, 2024-06-17:
https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#palm-os

Fr-Dae, 2024-06-17:
thank you, I've already looked at this documentation,
but there are no instructions on what to do, where to take the core, where to place it,

it's precisely because of the lack of instructions that I'm asking for help

AeolusUX, 2024-07-07:
I've seen you've already been helped by christian on the main repo so I'm closing this.

closed 2024-07-07 by AeolusUX