AeolusUX/ArkOS-R3XS #41: Another black screen

ByteCrusad3r, 2024-06-21:
hey how's it going. I had a small problem with a black screen appearing after the boot screen. However, I think the fault is not with the dtb files or the new screen, but rather with the new first boot script. I came to this conclusion after coming across a black screen and inserting the memory card into a computer and opening the partition manager, and finding a partition being allocated. Although the problem may seem silly, it can cause people to get confused that the problem may be somewhere else, but it was just the system allocating partitions without visual feedback.

AeolusUX, 2024-06-21:
There is no new first boot script it's the same as was before. And of course there wouldn't be visual feedback if you're not using the correct dtb file...

ByteCrusad3r, 2024-06-21:
II'm using the correct dtb for my screen, in this case panel 4, so much so that the boot screen appears, just the script doesn't give visual feedback.

AeolusUX, 2024-06-21:
It should, If it doesn't you probably have a bad flash or you haven't waited enough for it to run.

AeolusUX, 2024-06-21:
For your reference this is how it should look.
https://youtu.be/J-JCrmsoNWU?t=271

ByteCrusad3r, 2024-06-21:
I think the problem is not the flash problem, since I followed some recommendations from the Arkos developer, such as not using Balena Etcher, and instead I am using Raspbery Pi Image Writer, and if there were any problems with flash, it would show an error like it did with other images.

AeolusUX, 2024-06-21:
So this image being used by a lot of people has a problem that only you experience.... 
Is it really the image's problem? Idk.

If you however wanted help with this any further come to the discord or reddit.
Closing this as this is clearly not an issue.

ByteCrusad3r, 2024-06-21:
but regardless of the methods to write the image to the card, the script seems to work and allocate the missing partition. And it just doesn't give any visual feedback. I opened this issue just to fix this problem, especially because beginners in this world of custom firmware can be confused by a black screen that takes a long time, appearing on the screen after the boot image, making them think that the system is broken or there is a problem with the console itself.

ByteCrusad3r, 2024-06-21:
I'm also not sure if it's an image problem, and I really don't think it is. One of the reasons that comes to my mind could be some simple incompatibility, since I bought my r36s a few days ago and there may have been some small change in the hardware. Right now, I'm thinking about using another program to record the image, such as dd, as in the video you recommended.

AeolusUX, 2024-06-21:
Did it not display the same boot sequence on first boot?

ByteCrusad3r, 2024-06-21:
nope

ByteCrusad3r, 2024-06-21:
just a black screen appears, even with the script running

AeolusUX, 2024-06-23:
That's weird , I'll try with my panel 4 devices this weekend.

AeolusUX, 2024-07-07:
I could not replicate this issue on any of my cards and devices even with the newer 60hz dtb so I'm not really sure if this just happened to one of your installs because it's a bad flash or what. I'm closing this for the meantime as I think this is an isolated case.

closed 2024-07-07 by AeolusUX