AeolusUX/ArkOS-R3XS #42: Portmaster not running, another error

roo1234, 2024-06-23:
Have tried the fix mentioned here but no avail, I captured the flash fast error displayed when trying to run portmaster and it says: fileExistsError: errno17 harbourmaster.
Since last update. Also when connected to wifi.

AndresReyesDev, 2024-06-23:
I have the same issue!

AeolusUX, 2024-06-23:
I'm afraid I couldn't help you if the fix didn't work for you then it was not your problem. And that's the only problem that is reported and confirmed by people that I know of. Try portmaster discord if someone else is experiencing this because as far as I'm concerned there is no issue. I've setup dozens of devices with portmaster since I replaced the images that has an issue with portmaster but I haven't faced any issues on it since.

closed 2024-06-23 by AeolusUX