AeolusUX/ArkOS-R3XS #43: Arkos crashing after the WIFI connection

gabrielcupertino, 2024-06-25:
After one night scapping for game covers, my R36S just froze on a black screen (had the newest OS version installed). 

Every time I tried to boot it, it crashed after the first screen (on that Arakos black screen) and pressing down on the volume, appeared to me that it was something to do with the wifi connection, but I just couldn't put a finger on it, since the WIFI dongle wasn't connected and I had no 2nd SD cart inserted to minimize issues...
 
Nothing seemed to work: soft reset just gave me the same problem and I even tried to reflash the SD multiple times with a brand new Arkos install and nothing... (for caution, I tried various good brands SD cards, and nothing)

Since I was hopeless, tried the absurd: just disconnected the batery and... It came back to life! Problem is: every time I try to connect to the internet, it crashes again and I have to disconect the batery to reboot it (tried different adaptors and different dongles - same result).

Long story short: I'm stuck offline for now and the batery solution still puzzles me...

AeolusUX, 2024-06-25:
Seems to be common with the ones with built-in wifi. Have you tried with a different dongle/longer otg cable? Also if you already scraped your games what do you need the wifi for now?

AeolusUX, 2024-07-07:
Any updates on this?

gabrielcupertino, 2024-07-07:
> Any updates on this?

Yeah... Still nothing. Tried your solutions, but still, the wi-fi is freezing the OS, and it locks the boot routine when I try to hard reset it. The only solution is popping out the battery, and it boots normally.

Just for good measure, even updated the console with a fresh install of the latest OS and still nothing.

I'm a layman, but it seems to me that the boot is trying to find something related to the wi-fi, and then it gets stuck. At least it's what appears to be when I press volume down on the black boot screen. 

Usually, it doesn't take more than a couple seconds to go from the black Arkos boot to the Retroarch loading, but after the wi-fi problem, it's frozen, looping on a count down for something different each time.

AeolusUX, 2024-07-08:
If it happens even with a fresh install I think your hardware is the issue. This probably is another isolated case that I don't think updating the os would fix so I'm closing this.

closed 2024-07-08 by AeolusUX

jahed, 2024-07-23:
I had this exact same issue today. I was sharing my phone internet via USB tethering and then after a restart it showed the boot image and then a black screen. Thanks @gabrielcupertino for sharing the battery removal workaround.