AeolusUX/ArkOS-R3XS #45: After update, EmulationStation freezes at start in 2 SD card mode

janaxhell, 2024-06-29:
I've just updated ArkOS (P4 version) from the Update menu, all went well, except now each time I turn on R36S, EmulationStation shows the usual scan operation, but hangs at "Options" or some other entry nearby for about 2 minutes, then finishes and moves on to wheel menu.
This happens only in 2 SD card mode, but it  never happened with previous ArkOS version. 
Any fix?

AeolusUX, 2024-06-29:
Did you switch to main sd before updating?

janaxhell, 2024-06-29:
Of course. 
I have to add that yesterday I tried to do the update, but it kept repeating "You haven't typed OK", while I did, both in lowercase and uppercase. Todey I have tried again and it updated with no issue.

AeolusUX, 2024-06-29:
That's normal. I disable it when I don't intend people to update yet.

Try and set Start>UI Settings>Scan Ports folder on Boot is set to off.

janaxhell, 2024-06-29:
It is already set to OFF

AeolusUX, 2024-06-30:
Did you turn it off again?

> It is already set to OFF EDIT tried to set it ON = no go, switched back to SD1 mode = no go, again back to SD2 + read from SD1 and SD2 and it's the same, ES hangs for about 2 minutes at every boot.

janaxhell, 2024-06-30:
Of course, I've done systematic cross-tests of all combinations and each time I had to wait 2 minutes. 
Only thing I can add, but this was like that even with previous version, is that after adding a SD2 and switching to read from SD1 and SD2, all systems are shown twice in the wheel, despite the fact that I stored roms only on SD2. On SD1 I only have ArkOS

AeolusUX, 2024-06-30:
I just found out that this is already known by christian so if/when he releases a fix for it that should also be available to users of this fork.

janaxhell, 2024-06-30:
Oook, so I will receive a notification of the new release like I did for this one, right?

AeolusUX, 2024-07-07:
There hasn't been a new release yet but if you're able to please do an update via OTA as that should fix this issue.

closed 2024-07-07 by AeolusUX

janaxhell, 2024-07-07:
Oh, I didn't notice this news, it came at the same time of the ScummVM thread.
Yes! It works, thank you! I have updated and now no more ES stuck at boot.

Besides that, after updating, I have re-enabled "Read from SD2", but NOT "Read from both SD1 and SD2" and now I can see ScummVM from SD2 as it should be. But now I wonder if this is a spurious effect of what I did before here https://github.com/AeolusUX/ArkOS-R3XS/issues/48#issuecomment-2212395069

janaxhell, 2024-07-07:
UPDATE
I found out an interesting issue: I saw that ES said 124 systems available: that was because it was counting all duplicate systems from SD1 and SD2 (I have all on SD2). So I disabled from Visible Systems all those from SD1. Got out of ES Options, took a look around, then entered Options again: this time ES said "6 systems available" (which meant Options, Retroarch, Favorites, etc) and ALL the systems on SD2 were disabled as well. 

So, if you have all roms on SD2 and disable Visibility for the bogus systems on SD1, all the SD2 systems are disabled as well.
Either they are all enabled, or none.