AeolusUX/ArkOS-R3XS #47: No WIFI after update - MediaTek MT7601U [MI WiFi] VID_2717&PID_4106

amxcs, 2024-06-30:
Hello, after the update I no longer had Wifi - as if I had not inserted the USB dongle.
What could it be and is there a solution?

Device: R36S
Version of ArkOS before update: ArkOS 2.0 (02232024)
Version of ArkOS after update: ArkOS 2.0 (06272024)

AeolusUX, 2024-07-01:
There shouldn't be any issue from 02232024 to 06272024. Have you tried restarting the device while the wifi dongle is connected?Have you checked if your otg adapter is working properly?

amxcs, 2024-07-01:
The OTG adapter and dongle are working properly. I flash an older version twice - it works. After an update or flash with the latest version - no.
I tried restarting while the dongle was inserted - again it didn't work.

AeolusUX, 2024-07-01:
https://github.com/AeolusUX/ArkOS-R3XS/issues/19#issuecomment-2061297847
Could you try with this?

amxcs, 2024-07-01:
This worked, thanks!

luzfcb, 2024-07-03:
@AeolusUX Even after [apply this backup  ](https://github.com/AeolusUX/ArkOS-R3XS/issues/19#issuecomment-2061297847), the network still does not work with my `MT7601U` and `RTL8188EUS` .

My OTG adapter works as expected because I am able to mount a FAT 32 USB Flash drive

Is there any way to acess a shell on that device?

```
ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

AeolusUX, 2024-07-07:
> @AeolusUX Even after [apply this backup ](https://github.com/AeolusUX/ArkOS-R3XS/issues/19#issuecomment-2061297847), the network still does not work with my `MT7601U` and `RTL8188EUS` . I'm using the `Original Screen Updated (06/30/2024)` version .
> 
> My OTG adapter works as expected because I am able to mount a FAT 32 USB Flash drive
> 
> Is there any way to acess a shell on that device?
> 
> ```
> ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
> ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
> ```

You can using ssh but you don't have wifi so that's a no go. That fix is for previously working dongles on older images. Are your dongles working on older images?

amxcs, 2025-01-01:
Hello,
on version (ArkOS 2.0 11272024 Updated 12/06/2024) it still doesn't work.