AeolusUX/ArkOS-R3XS #48: ScummVM 2.8.1 not working on ArkOS 2.0 2024-06-27 build on R36S

janaxhell, 2024-07-02:
I have a full 2.8.1 set on my Win PC and it works fine. I have enabled ScummVM visibility from ES UI settings on R36S. I have copied some games to \roms2\scummvm\ and tried to BOTH add a .scummvm files with relative game ID inside and outside each game's folder. No game was detected in either way and no ScummVM entry appears on the ES wheel. Tried the Retroarch "Load Core" method, but as soon as I add/load a game, ScummVM exits to Retroarch. 
Obviously everything else works fine, only ScummVM refuses.
After asking on /r36s I've been invited to open an issue about this.

janaxhell, 2024-07-06:
UPDATE
After tinkering for a while I found out that IF \scummvm folder contains some games (NOT necessarily the same games) in BOTH SD1 and SD2 AND it is enabled on both, it appears twice in the wheel, with the one referring to SD1 showing only \roms games and the one referring to SD2 showing both \roms+\roms2 games.
If one of them is disabled from the ES UI settings, the other gets automatically disabled too and none of them appears in the wheel.
So it's mandatory that BOTH are enabled AND contain at least some games.

AeolusUX, 2024-07-07:
So why do you have scummvm games on both cards? I'd be sure to disable that for the next update.

janaxhell, 2024-07-07:
I don't have them on both because I want, but because that's the only way to make ScummVM entry appear in the ES wheel.

AeolusUX, 2024-07-07:
Which sd are your scummvm games at?

janaxhell, 2024-07-07:
On both, but the main collection is on SD2. On SD1 I just put a few to make it appear in the wheel. They are like a placeholder.

closed 2024-09-06 by AeolusUX