AeolusUX/ArkOS-R3XS #49: Microvision emulation display is cut off on R36S

ZoranLain, 2024-07-03:
On R36S (I have the original display panel), Microvision emulation has a display issue. When I start a game (any rom), the display seem to be off. The screen seem to be cut off. It is as if the resolution is wrong.

AeolusUX, 2024-07-03:
Really? Can you show an image? If resolution is wrong does it even matter for these kind of games?

ZoranLain, 2024-07-03:
Sure.  Here are 2 photos.  You can see the play area is off to the side and go out of bound.   I also have the same game running on my old RG351P on ArkOS (wuMMLe fork).  That's what it supposed to look like. 
[image]
[image]

AeolusUX, 2024-07-26:
Ok just got time to try and fix this. I found the issue and will add it for the next release. Was not able to find a proper .bmp and _snap.bmp for my device so I just tested with it bare xD

AeolusUX, 2024-08-05:
Should be fixed with latest update.

closed 2024-08-05 by AeolusUX