AeolusUX/ArkOS-R3XS #50: R36S - Logo/Black Screen/

NNC007, 2024-07-03:
Hi everyone, I’m hoping someone might be able to help me. I’m experiencing an issue similar to others. First of all, my console is brand new. It’s currently running on the stock card with ArkOS 04242024, but I want to replace it with a new one. As we all know, we can’t predict how long the stock one will last. For this reason, I bought two Samsung Evo Plus red 256GB cards. I’ve downloaded several OS versions and tried a few different things, but so far, nothing has worked. Please see attached photos
[image]
[image]
https://streamable.com/t4450c - and that looks like all the time .
Thanks in advance 
Matt

AeolusUX, 2024-07-07:
Have you tried with your stock one from your stock cards? and are these tests (replacing dtb files) done after a fresh flash each time?

AeolusUX, 2024-07-12:
Any update on this?

NNC007, 2024-07-12:
> Any update on this?

Hi [AeolusUX]

I apologize for the delayed response. I would like to express my gratitude for your interest. However, I have already found a solution to my problem. It was an issue with the SD card (Samsung Evo Plus Red 256GB), which appears to be incompatible with the R36S.

Best regards.
Matt

closed 2024-07-12 by NNC007