AeolusUX/ArkOS-R3XS #54: NDS ART

Healer101015, 2024-07-18:
I made some artwork for NDS, do you think it would be cool to put them in the project's optional?

AeolusUX, 2024-07-24:
Can you share them here?

Healer101015, 2024-08-02:
SURE
[image]
[image]

AeolusUX, 2024-08-02:
Looks nice do you have more? Could you share them as .zip files so I could upload them for people who wants them?

Healer101015, 2024-08-02:
So far I have these
[all arts.zip](https://github.com/user-attachments/files/16474202/all.arts.zip)

AeolusUX, 2024-08-05:
Great I'll make a new repo for it and include the ones I've modified.

AeolusUX, 2024-08-09:
https://github.com/AeolusUX/arkos-ndsoverlay-gallery

closed 2024-08-09 by AeolusUX