AeolusUX/ArkOS-R3XS #56: All download sites netdisk were blocked in China, So is there any http download link for the imgs?

whmzsu, 2024-08-05:
All download sites netdisk were blocked in China, So is there any http download link for the imgs?

AeolusUX, 2024-08-05:
Would https://filebin.net/ work for you?

whmzsu, 2024-08-06:
This site seems OK ,   I just tested it ,  by  uploaded and downloaded  a smaill file .  Thanks very much.

AeolusUX, 2024-08-07:
Sucks that it only works for a week and it gets deleted. Which specific image do you need btw?

whmzsu, 2024-08-08:
Is those imgs' difference only the dtb file ?  If it is , either one is OK.

whmzsu, 2024-08-08:
And if could ,  upload to this site   https://archive.org/

AeolusUX, 2024-08-08:
I uploaded them to github on the release you just need to extract them to combine the 001 and 002 files.

whmzsu, 2024-08-08:
> I uploaded them to github on the release you just need to extract them to combine the 001 and 002 files.

Nice.