AeolusUX/ArkOS-R3XS #57: FN+ Up/Down changes both save state slot(only up) and brightness & visual indication for volume.

jpelizza, 2024-08-06:
Hi,

On SNES core(retroarch/snes9x) using FN+Up:
FN+Up changes brightness and save state slot.
FN+Down only changes brightness.

I'd propose to remove changing state slot select to FN+Left/Right so it falls in line with mednafen emulators.
As for volume indication, there is none, if it's possible to add it'd be appreciated.

I'm using R36S 4p screen in case that's relevant.

AeolusUX, 2024-08-16:
I don't understand isn't change state slot on Select + Left/Right?

`Select + Dpad Left: Decrement selected save state slot
Select + Dpad Right: Increment selected save state slot`

cmartell-at-ocp, 2024-09-04:
Surprisingly was seeing the same thing on my side for GBA, but figured out that it was because I had never reset the Retroarch defaults. After doing so from tools, it moved my hotkey from FN to Select.

That being said, even after doing that, Select + UP would decrement save state slot, but Select + DOWN would do nothing until I manually added it to the hotkeys in Retroarch (wasn't bound to anything).

Select + Right / Left did nothing for me.

fregapple, 2024-09-09:
I wouldn't say this is an issue. You just need to change the hot keys for save states within retro arch

closed 2024-09-16 by AeolusUX