AeolusUX/ArkOS-R3XS #58: R36S - retrorun32 on Atomiswave and Naomi

ZacharyFoxx, 2024-08-10:
The retrorun32 and flycast-rumble32 combination works wonders on the R36S when running Dreamcast/Naomi/Atomiswave, but for Naomi and Atomiswave I think the button layout is messed up.  Games like sfz3up have the punch buttons on Y, B, A and the kicks on L,X,R. 

In other games like CvsS, the punches are on B and A, a kick on X and another on R. I believe this is related to mapping the layout of retropad32, Retroarch's virtual control.

Is there a way to adjust or allow someone to create a remap file for retrorun32?

AeolusUX, 2024-09-19:
Iirc retrorun needs to be recompiled to change button mappings that's why it's not in my current priority list. I also don't play any atomiswave or naomi so I have no idea of what the correct mappings should be. If you have any suggestion for proper mapping for them maybe you can give me an idea and I'll try and make a build for retrorun that fits that.

AeolusUX, 2024-10-02:
Any ideas for this?

ZacharyFoxx, 2024-10-04:
> Any ideas for this?

Hello. Sorry for not having answered earlier, because unfortunately my sdcard that I use in the R36S got corrupted and I will have to reinstall everything.

I will do that this weekend and report back here.

But I think this topic can help give an idea of ​​the layout problem in the atomiswave and naomi games:

https://www.dreamcast-talk.com/forum/viewtopic.php?t=14042&sid=b53516abd7c74499c1ffd5ce03d242ab

and this one:

https://www.dreamcast-talk.com/forum/viewtopic.php?t=14042&start=10

Some games have a completely weird button layout and when you bring it back to the traditional layout it looks weird.

I will reinstall the R3XS this weekend and post here what could be done.

But the most likely thing would be to leave one core for the Dreamcast and one for Naomi/Atomiswave due to the layout.

ZacharyFoxx, 2024-10-04:
Just another comment.

Looking back at what I wrote about the layout of the sfz3up game, it seems that it is using the direct configuration of a 6-button Sega Genesis/Sega Saturn controller that Retroarch uses:

https://docs.libretro.com/library/genesis_plus_gx/#joypad

It's a weird layout when you translate the Genesis/Saturn controls to the layout of a PS2/X360 controller, so it looks like this:

Sega Saturn -> X360
         A           ->     X
         B           ->     A
         C           ->     B
         X           ->     L1
         Y           ->      Y
         Z           ->      R1

If the retrorun32 and flycast-rumble32 combination uses the Retroarch layout, that might explain the messed up layout.

I'll do a series of tests over the weekend and report back here.

ZacharyFoxx, 2024-10-07:
To check what the reconfiguration would be, I tested via retroarch32.

Considering what you said, the solution would be a retrorun32 core with the following button configuration:

B - btn4, 
A - btn5, 
Y - btn1, 
X - btn2, 
R - btn6, 
R2 - btn3, 
L - btn8, 
L2 - btn7

keeping all the others.

With this, 6-button games like SFZ3UP, CVS2, MvsC have the same configuration of weak and medium buttons in Y, X, B, A as on a snes controller, with the strong punch on R2 and strong kick on R. 5-button games like Hokuto no Ken use the Y, X, B and A buttons like in SF2 on the snes and use R2 for special defense. And others like Guilty Gear have the correct arcade configuration: B as punch, Y as kick, X as slash and R2 as strong slash.

It's the best possible approach.

raay8456, 2025-02-14:
Please make a ui for retrorun, it's best for Dreamcast

GazousGit, 2025-04-27:
I'd like to add a request on this issue, retrorun32 use 'Start' button to quit and it's realy really annoying because when you start a dreamcast game you have to press start several times in the menus until you access the game so you have to wait every time for the popup to go until you can press start again otherwise you'll end up quitting retrorun32.

It would be great if the quit button would be set to 'FN' instead of start or even select (since Naomi and atomiswave will use select for coins)

GazousGit, 2025-05-11:
@ZacharyFoxx could we discuss this over discord ? I'd like to create a list of all games requiring a change of input (or a rotation of screen, it can be done through retrorun menu, but would be great if it was already rotated by default when you lanch the game)