AeolusUX/ArkOS-R3XS #60: Switching the SD2 for ROMS

gallardworth, 2024-08-16:
I cannot switch to SD2. Black screen appears with no word like swapfile or something then it goes to normal, changes nothing. In 351files, I look up to ROMS2 file, in that file my game files don’t appear but nds file. I formatted both my SD cards. I installed back all of it. It switched once to SD2, then it hasnt since then. What should I do? I can format cards one more time but it is frustrating. Thank you in advance.

AeolusUX, 2024-08-16:
Firstly this is not for tech support. Consider going into the r36s discord channels of retrohandhelds or retrogamehandhelds or the R36S subreddit.
Are you sure the sd2 is working and compatible on that slot? If the script didn't do anything and roms2 didn't mount it's likely that it's not a supported card. Check with a different card.

closed 2024-08-16 by AeolusUX

Reutertu3, 2024-08-19:
It appears to be a visual bug. After switching the menu entry doesn't update, but upon reboot the menu entry shows the expected behavior.