AeolusUX/ArkOS-R3XS #63: R36S Original vs Type4 Hardware Differences

6040, 2024-09-15:
Hi
this is a question rather than an issue.
I have four R36S consoles. Two with "Original Screen" and two with "Type 4".
My goal is to play the original Gameboy Tetris on two consoles against each other via Netplay.
This works fine with the "Type 4" Consoles. The emulator does not matter but both mGBA and TGB Dual work. But it does not work on the original ones.
I use the same TP-Link Dongle on all the consoles and use the same version of Arkos and rom on all. 
It says compatible content found and then the activity light on the wifi dongle blinks rapidly and never stops. It never lets me start the two player option.
On the Type 4 consoles the wifi dongle blinks briefly and then it just works.
Any help is gladly appreciated
6040

AeolusUX, 2024-09-16:
Unfortunately I don't have two consoles with original screen anymore so it's very hard to have this replicated. Also the only thing replaced from type 4 and original screen consoles are screen so this might be something about the screens themselves. Have you tried playing with one original screen vs one panel 4 device?

6040, 2024-09-17:
I tried to mix old vs panel 4 like you suggested, but this does not work either. Same behavior than old vs old. Could this display driver affect the wifi driver in some way? Which logs would I need to look at to get a hint of why it does not work?

AeolusUX, 2024-09-17:
I know that there are issues on wifi with panel 4 but not the other way around so this seems weird. Maybe you could try other dtb files like the original ones that your device came with, you could also try ones from this repo https://github.com/tech4bot/r35s

6040, 2024-09-19:
I tried tech4bot and the original dtbs. Both work for the os (The original is missing the FN button) Unfortunately Netplay with Tetris still does not work. I also tried AmberElec but it does not work either. Are there any log files that would help me find this issue?

AeolusUX, 2024-09-19:
Not that I know of. Basically as these are clone devices themselves some of their parts are low quality like the screens are clones of clones so they're very bad that sometimes they mess with other things as they are unstable. Would be easier to get all panel 4's instead for your use case.

6040, 2024-09-21:
Thanks
you are probably right and I should get new consoles, jJust have to convince my godchild to swap them out.

closed 2024-10-02 by AeolusUX