AeolusUX/ArkOS-R3XS #65: R33S hot keys not working on stand alone emulators

yuda00-1, 2024-09-28:
Hotkeys using R3 or any analog buttons cannot be opened when using R33S due to the lack of analog buttons. Menus don't open in stand alone PSP and DS emulators hence game also cannot be exited. cannot also swap screen position on DS dual screen layout

PSP (not minis) also does not appear in emulation station (it appears on stock os) and can only be accessed via retroarch

AeolusUX, 2024-10-02:
Yeah I might be updating it for this months release. I haven't been updating this much since there are only a few R33S before but now that they released it I might be doing updates for it soon.

yuda00-1, 2024-10-06:
Please do. The R33s is a really good handheld

AeolusUX, 2024-10-22:
This has been taken into account on the latest update. There will also be a single image for both the R33S and R35S/R36S from now on.

closed 2024-10-22 by AeolusUX