AeolusUX/ArkOS-R3XS #67: RTL8188FU WiFi module missing in kernel

nolt, 2024-10-04:
Hello I have few WiFi adapters based on RTL8188FU which is missing in ArkOS and mine WiFi dongle doesn't works. Could you add this module into ArkOS.
https://github.com/kelebek333/rtl8188fu

Mine device is: R36s
ArkOS version: latest

PS. I was redirected to your repo from https://github.com/christianhaitian/arkos/issues/1197 cuz he do not support R36s. IMHO its odd cuz mine request wasn't for exact device but kernel module.

AeolusUX, 2024-10-04:
We already have that afaik. Are you using the community image?

AeolusUX, 2024-10-04:
If you're using the RG351MP image you could just add this file into

The kernel modules directory at `./usr/lib/modules/4.4.189/kernel/drivers/net/wireless` and then do `sudo depmod`

[rtl8188fu.zip](https://github.com/user-attachments/files/17260322/rtl8188fu.zip)

closed 2024-10-04 by AeolusUX