AeolusUX/ArkOS-R3XS #68: R35S screen is cut off on the bottom

xactactx, 2024-10-08:
When I first got the R35S and I was installing ArkOS images onto it, the screen was aligned correctly, but since I flashed one of the newer versions of ArkOS, the screen is misaligned vertically and the bottom of the screen is cut off. This is happening with the newest release of ArkOS (08232024-1).

Two people have reported the same issue on Reddit:

https://www.reddit.com/r/SBCGaming/comments/1fumx1y/r35s_screen_not_aligned_vertically_on_arkos/

https://www.reddit.com/r/SBCGaming/comments/17cs2rr/r35s_screen_cut_off_at_bottom/

AeolusUX, 2024-10-16:
You just go to Options>Advanced>Screen - Switch to Tony Screen Timings

closed 2024-10-16 by AeolusUX

xactactx, 2024-10-16:
Thank you, I hadn't seen the "switch to Tony timing" option, I clicked on it and my R35S rebooted, but unfortunately the screen is still cut off on the bottom. I wonder if other people can solve the problem with Tony timing or not. On my device it doesn't seem to make any difference between Anbernic and Tony timing.

AeolusUX, 2024-10-17:
> Thank you, I hadn't seen the "switch to Tony timing" option, I clicked on it and my R35S rebooted, but unfortunately the screen is still cut off on the bottom. I wonder if other people can solve the [...]

That's weird, could you try to replace your dtb files with these?

[R35S-DTB.zip](https://github.com/user-attachments/files/17411313/R35S-DTB.zip)

Just paste them over on your boot folder and overwrite these files.

xactactx, 2024-10-18:
Thanks for the follow up, unfortunately it didn't make a difference, the screen is still cut off, either with original timings or Tony timings. It looks just like the two Reddit threads that I posted.

AeolusUX, 2024-10-23:
Would you be able to source your old R35S dtb that doesn't cut the bottom and doesn't have the fn button? If you have a copy of your old one somewhere we could check.

xactactx, 2024-10-23:
I don't have access to the original versions of ArkOS that were being made in 2023, but I remember that those versions worked fine

AeolusUX, 2024-10-24:
Does that work as is or you have to use tony timings as well?

xactactx, 2024-10-24:
I'm sorry I don't understand, IDK where to download the old versions of R35S that didn't cut off the screen, but I'm pretty sure they it worked fine before