AeolusUX/ArkOS-R3XS #70: bluetooth support?

Lana-chan, 2024-10-10:
hi! i'm trying to get bluetooth working on a R36S handheld with a dongle on OTG. it's supported and working, i can use `bluetoothctl` on the terminal to scan for devices, so that's all good. but there are no bluetooth options anywhere through the GUI, on emulationstation or the options folder, like there is on other devices. i tried to look for scripts to add myself but didn't really find anything, is there any chance those could be added to the R36S release?

Distro: ArkOS 2.0 (08232024-1)(AeUX)
EmulationStation V2.13.0.0

thanks!

AeolusUX, 2024-11-27:
There is a utility made by christian for the 3566 devices, I'll check if I can add that to this build, if it is would you be able to test it as I don't have any bluetooth device(dongle) that I can test with this.

Lana-chan, 2024-11-29:
yep, i can test it with bluetooth audio and two controllers (one generic, one xbox)

VolodymyrDA, 2025-02-09:
My wifi asus adapter that working fine on arkos rs36 is not working there :(

bicksvaporub, 2025-04-12:
> There is a utility made by christian for the 3566 devices, I'll check if I can add that to this build, if it is would you be able to test it as I don't have any bluetooth device(dongle) that I can test with this.

please continue this, this is what will make this firmware complete package. thanks

EatPrilosec, 2025-05-02:
bluetooth support would be greatly appreciated, and the only reason im even considering another OS

mav2010, 2025-06-12:
I second this. I would greatly appreciate bluetooth support!

mule-bya, 2025-08-16:
Yeah, bluetooth would be amazing! I'd like to connect a keyboard for some Portmaster games, but all of my spare keyboards are bluetooth.