AeolusUX/ArkOS-R3XS #72: Sugestion: Power button for shutdown

CBearr, 2024-10-28:
Hi, maybe you could add an option to "gently" shut down the handheld (R36s) by a double tab at the power button?
Background: we could use this shortcut like if you are in the emulator station and use "quit - shutdown system".

I think long press power button is force shutdown directly. Or use the reset button for quit - shutdown system?

Reset and force shutdown could end in data loss. No one need the reset button. In our German community some people use this to go back from the game to the emulator station (-. -)

AeolusUX, 2024-10-29:
It's already been there :)
FN+Power should do what you want.

closed 2024-10-29 by AeolusUX

CBearr, 2024-10-30:
Thanks! But it shutdown much more faster that does if I shutdown it from the settings menu... and no black screen with the version number and the 4 dots blinked in white red. Really the same way?