AeolusUX/ArkOS-R3XS #82: NDS default backgrounds

captainbk, 2024-12-30:
Could you make the NDS emulator not default to having a background in dual screen mode. I've seen multiple reports of confusion, and how to change it doesn't appear to be easily found online.

Dattebayooooo, 2025-01-01:
Any idea how to change the ds core to melonDS?

AeolusUX, 2025-01-13:
> Could you make the NDS emulator not default to having a background in dual screen mode. I've seen multiple reports of confusion, and how to change it doesn't appear to be easily found online.

I'll figure out a way for that maybe a script to disable NDS Overlays.

> Any idea how to change the ds core to melonDS?

That's not gonna be available on the RK3326 because it runs like ass.

closed 2025-01-13 by AeolusUX