AeolusUX/ArkOS-R3XS #84: Webinterface broken after update to 12242024

heiq, 2025-01-16:
The Webinterface is not working anymore since i updated. After a wifi update from 11272024 i did a fresh flash but got the same issue.

beicom, 2025-02-02:
Same happens on my device, solved by connect over ssh and add permission to run filebrowser.

sudo chmod +x /usr/local/bin/filebrowser

====
ark@rg351mp:~$ sudo /usr/local/bin/filebrowser
sudo: /usr/local/bin/filebrowser: command not found
ark@rg351mp:~$ sudo /usr/local/bin/filebrowser
-bash: /usr/local/bin/filebrowser: Permission denied
ark@rg351mp:~$ ls -la /usr/local/bin/filebrowser
-rw-r--r-- 1 root root 16515224 Oct  3 21:16 /usr/local/bin/filebrowser
ark@rg351mp:~$ sudo chmod +x /usr/local/bin/fi
ark@rg351mp:~$ sudo /usr/local/bin/filebrowser
2025/02/02 08:56:30 Warning: filebrowser.db can't be found. Initialing in /home/ark/
2025/02/02 08:56:30 Using database: /home/ark/filebrowser.db
2025/02/02 08:56:30 No config file used
2025/02/02 08:56:30 Listening on 127.0.0.1:8080

AeolusUX, 2025-02-06:
> Same happens on my device, solved by connect over ssh and add permission to run filebrowser.
>
> sudo chmod +x /usr/local/bin/filebrowser
>
> ```
> ark@rg351mp:~$ sudo /usr/local/bin/filebrowser
> sudo: [...]

I see so it's permissions was the issue thanks for that I'll add a fix on the latest one.