AeolusUX/ArkOS-R3XS #85: Emulationstation takes more time to load after latest update 12242024

rola45, 2025-01-16:
I updated from the device, everything is ok but the following power-ons or reboots make Emulationstation take about 15 seconds on its own to load, making it a total of 30 seconds (before it took 15secs) I have dual 128gb ADATA premier sd installed, before updating I activated the option to use the main sd card for roms, then after update, reactivated for 2nd slot. Should I format the second SD? I already restored a previous ark os backup but the problem continues

grumpy2msc, 2025-01-17:
Same Issues for me as @rola45. I updated from the 09292024 to 12242024. Before, start-up took all of 12 seconds. It takes close to 1 minute now to get in and be able to scroll through the menu. I am running dual SD cards, 16gb and 128gb. I tried doing it on a single SD card (the 128gb) and it still boots up incredibly slow.

grumpy2msc, 2025-01-17:
Well as for now my fix was to go back to 09292024. I do not see a difference in the actual emulation so for now it is what it is. What would be the advantage to use the 12242024?

grumpy2msc, 2025-01-18:
> I updated from the device(11272024 to 12242024), everything is ok but the following power-ons or reboots make Emulationstation take about 15 seconds on its own to load, making it a total of 30 seconds (before it took 15secs) I have dual 128gb ADATA premier sd installed, before updating I activated the option to use the main sd card for roms, then after update, reactivated for 2nd slot. Should I format the second SD? I already restored a previous ark os backup but the problem continues

Ok after some testing. How did you install the 12242024? was it from the download? or did you upgrade from the device to it? I reverted back to the 09292024 because it started up faster. Instead of installing the 12242024, I upgraded to it from the device, and it starts up much faster. It has the modifications I made to it and doesn't install the extra folders like screen 3 and screen 4.

rola45, 2025-01-18:
> Ok after some testing. How did you install the 12242024? was it from the download? or did you upgrade from the device to it? I reverted back to the 09292024 because it started up faster. Instead of [...]

Installed via upgrade option on the device itself, i just figured that ports folder is the problem, takes a lot of time to load, deleted the games inside and loads faster, i think im going to reflash the 11272024 image

rola45, 2025-01-18:
Update: reflashed to previous version (11272024) everything working perfect, 16secs loading time with 100gb+ rom library.

closed 2025-01-19 by rola45