AeolusUX/ArkOS-R3XS #87: R36S PSP Emulator is not visible

SarpDervish, 2025-01-17:
Hello, I'm new at this. I bought the R36S console and installed arkOS. Although there is a file called psp in the Roms folder and there are files like ppsspp.ini, I can't see the emulator on the console in any way. I downloaded 1-2 .iso files, but since there was no file called games in it, I created it myself and threw these .iso files in it, but I still can't see the psp emulator. When I look with retroarch, it says that the core file has already been downloaded. I'm using the latest version of arkOS. I would appreciate your help.

grumpy2msc, 2025-01-18:
You shouldn't need to make any other folders. just put the .iso file (unzipped) in the psp folder.

[image]

SarpDervish, 2025-01-18:
thank you! I can play it now

closed 2025-01-18 by SarpDervish