AeolusUX/ArkOS-R3XS #88: R36S ArkOS no longer recognizes my second SD card

AleSan78, 2025-01-18:
ArkOS no longer recognizes my second SD card since the 12242024 update

rola45, 2025-01-18:
I updated from 11272024 to 12242024 via OTA and several problems arose, my second SD did recognize it but it took a lot of time to just load emuationstation, I had to format and the problem continues, I think it is the version of the image, I will reflash to 11272024 that version worked well for me, neither recommend use the boot logo video option, takes more time to load

rola45, 2025-01-18:
Update: reflashed to previous version (11272024) everything working perfect, 16secs loading time with 100gb+ roms.

AeolusUX, 2025-01-19:
Did you guys read the warning prior to updating?

AleSan78, 2025-01-20:
No, what warning, I didn't see

closed 2025-01-20 by AeolusUX