AeolusUX/ArkOS-R3XS #90: Scummvm broken 12242024 !!!

THBR-ctrl, 2025-01-25:
The scremmvm is broken after the 12242024 update. The games are not read, when it creates an executable file it recognizes it but does not execute gamedata. I tested it in the game Grim fandango, my friend tested it in the 11272024 version and worked perfectly, but everyone who is in the 12242024 version has this defect of loading.

https://github.com/user-attachments/assets/1d70b862-8e42-435c-b72a-387cc5fd7a11

[image]
[image]

Flah438, 2025-01-26:
American Laser Games same problem too, thnx

AeolusUX, 2025-01-30:
> The scremmvm is broken after the 12242024 update. The games are not read, when it creates an executable file it recognizes it but does not execute gamedata. I tested it in the game Grim fandango, my friend tested it in the 11272024 version and worked perfectly, but everyone who is in the 12242024 version has this defect of loading.
> 
>  grom.fandango.mp4 
> [image] [image]

Try this

Take the attached compressed file and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

Once completed, reboot the device.

Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Tools, hit A, then scroll all the way down to scummvm_fix

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18605476/arkosbackup.tar.gz)

., hit A.

The unit will reboot again.

Start testing your scummvm to see if there's any difference.

THBR-ctrl, 2025-01-31:
> > The scremmvm is broken after the 12242024 update. The games are not read, when it creates an executable file it recognizes it but does not execute gamedata. I tested it in the game Grim fandango, my friend tested it in the 11272024 version and worked perfectly, but everyone who is in the 12242024 version has this defect of loading.
> > grom.fandango.mp4
> > [image] ![Image](https://private-user-images.githubusercontent.com/196480907/406688671-c924677a-92f1-4d6d-b5c8-336b45614e05.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzgyOTIzNDEsIm5iZiI6MTczODI5MjA0MSwicGF0aCI6Ii8xOTY0ODA5MDcvNDA2Njg4NjcxLWM5MjQ2NzdhLTkyZjEtNGQ2ZC1iNWM4LTMzNmI0NTYxNGUwNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEzMVQwMjU0MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYWE2NzZmNjMzMGIwYjNmYjYzNDM5YTdhYzlhODAzNDE4Yzc1MzgwN2U5NDI4NDQ0Y2FkYWMxNDA4YzU5Mjg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.1rk60-URtJ5uRS7atA0bITlaaQ8TN7wV9S7QwqMoaWw)
> 
> Try this
> 
> 1. Take the attached compressed file and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.
> 2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.
> 3. Once completed, reboot the device.
> 4. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Tools, hit A, then scroll all the way down to scummvm_fix., hit A.
> 5. The unit will reboot again.
> 6. Start testing your scummvm to see if there's any difference.
> 
> [arkosbackup.tar.gz](https://github.com/user-attachments/files/18605476/arkosbackup.tar.gz)

It worked great. Thanks 👌😁

closed 2025-01-31 by THBR-ctrl