AeolusUX/ArkOS-R3XS #93: R36H ARKOS som roms are not working Atari7800 and Vectrex

michel130, 2025-01-31:
Hello,

Some  ROMs are not working on the Atari 7800 and Vectrex. These ROMs fail to start.
For Vectrex, some ROMs remain stuck on a black screen.
For the Atari 7800, the system stays on the Atari logo and then stops.

michel130, 2025-02-05:
The Atari 7800 is working for me now; the issue was related to NTSC and PAL games.

The only problem I haven’t been able to solve is with the Vectrex.
When I start a Vectrex game, it boots up with sound and white text, but as soon as I try to play, the screen goes black, and nothing is displayed.

Hopefully, someone can help me out with the Vectrex issue.

AeolusUX, 2025-02-06:
This is not really tech support unless this is a bug, you'd get better help and results in asking this on the retrohandhelds discord or the r36s subreddit where people have experience making these emulators work.

closed 2025-02-06 by AeolusUX