AeolusUX/ArkOS-R3XS #95: R36S-the joystick is reversed under arkos

hukaka22, 2025-02-04:
I updated the latest version of arkos(ArkOS R3XS V2.0 (12242024)). After entering the system, I found that the R-joystick was reversed up and down, and the L-joystick turned left and right into up and down. There is no remapping of the controller found in the system settings. Is there any way to solve this problem?

AeolusUX, 2025-02-06:
You might have a clone you just remap it depending on which system it is you're trying to run.

hukaka22, 2025-02-06:
> You might have a clone you just remap it depending on which system it is you're trying to run.

Thanks!How to remap it?

AeolusUX, 2025-02-06:
> > You might have a clone you just remap it depending on which system it is you're trying to run.
> 
> Thanks!How to remap it?

Depends on which system. It would really help if you add more info as it's hard to guess really.

hukaka22, 2025-02-06:
[image]

This, ArkOS R3XS V2.0 (12242024)

AeolusUX, 2025-02-06:
You mean in emulationstation?

hukaka22, 2025-02-06:
> You mean in emulationstation?

Yes，I can’t find where remap

VolodymyrDA, 2025-02-09:
i have updated version like your one, other with clean install from last image. All directions working fine

AeolusUX, 2025-02-26:
> > You mean in emulationstation?
> 
> Yes，I can’t find where remap

Is this a Y3506 board? A.K.A. Soysauce console" "酱油机"

Asterosaur, 2025-03-06:
I got the same issue as the author. 

>If so, try with this DTB : [rk3326-r36ss-linux.zip](https://github.com/user-attachments/files/18980546/rk3326-r36ss-linux.zip).

this file works exactly like one provided by the vendor

> 1. Go to Options > Advanced > Reset EmulationStation Controls

the app says press a to continue. But doesn't react to anything except VOL- button.

AeolusUX, 2025-03-06:
> I got the same issue as the author.
>
>
> this file works exactly like one provided by the vendor
>
>
> the app says press a to continue. But doesn't react to anything except VOL- button.

Make sure you edit boot.ini entry to use this new dtb file.

closed 2025-03-06 by AeolusUX

Asterosaur, 2025-03-06:
> Make sure you edit boot.ini entry to use this new dtb file.

I did it, still no effect. It boots up just as usual

AeolusUX, 2025-03-06:
> > Make sure you edit boot.ini entry to use this new dtb file.
> 
> I did it, still no effect. It boots up just as usual

What do you mean of course it boots as usual. This fixes the inverted up and down on emulationstation. Nothing else.

Asterosaur, 2025-03-06:
> This fixes the inverted up and down on emulationstation. Nothing else.

I mean it doesn't affect input at all

this how it looks after I edited boot.ini for new dtb file

[image]
[image]
[image]
[image]

AeolusUX, 2025-03-06:
> > This fixes the inverted up and down on emulationstation. Nothing else.
> 
> I mean it doesn't affect input at all
> 
> this how it looks after I edited boot.ini for new dtb file
> 
> [image] [image] [image] [image]

I just tried on mine and it worked. This fix was made because I've gotten a batch of this "soy sauce" r36s and this fixed the issue for me. I also know someone I've helped on the facebook group and one on discord that tried the same thing and it worked for them so it's probably your device or something is wrong with what you're doing.

hukaka22, 2025-03-07:
> I got the same issue as the author.
>
>
> this file works exactly like one provided by the vendor
>
>
> the app says press a to continue. But doesn't react to anything except VOL- button.

Thanks,I tried it too, but the button didn't respond.“ press a to continue. But doesn't react to anything except VOL- button.”

shotimer, 2025-03-15:
Hi, I experience the same. I cannot press "A" or "B" to continue after launching Reset EmulationStation Controls. Basically all buttons are not not responding except Vol down button. My joysticks are also inverted in Menu.

gr33k, 2025-03-24:
I have this issue as well for a R36S similar to what is pictured (if not the same). I used the original DTB as none of the 5 screen types would boot. Emulation station, ScummVM among a few others have joycons inverted or flat out wrong direction. I was able to remap it in Retroarch for SOME emulators, but I can't do it for emulation station ScummVM yet. Any suggestions on what to do to remap joycon? I use emulationstation on my RasPi and it has a controller settings option in the settings. From that - it lets you setup any controller simply...however this feature is not in ArkOS's settings menu to do so. Any way to get this back so people can easily remap the controls (i.e. under advanced settings or otherwise)? Thank you!

gr33k, 2025-03-25:
I noticed this issue is "Closed" but I am not sure I can find what the solution was.

Feature change - for that controller reset tool. Could you make it so no key press is required to reset. Perhaps a countdown with an option to press a key to cancel vs requiring a confirmation to proceed (as the keymaps not working will prevent folks from being able to confirm).

AeolusUX, 2025-03-25:
The solution was to use the soy sauce console DTB files. If you encounter any issues with mappings, note that adjustments can’t be made directly on the image since this is a clone, making it the only device with this specific problem. You’d need to remap the configurations yourself. Unfortunately, I no longer have one of these devices, so I can’t assist with the remapping process. I suggest reaching out to someone who owns the same device for help.

gr33k, 2025-03-25:
Thank you for your reply @AeolusUX - Could you please point me to the file/directory (or if there is a link/wiki on this) that would need to be modified and it's location? I'm happy to tinker and see if I can fix this. If so, I'd be happy to also share the file here for anyone else having the issues as well! 🍺

gr33k, 2025-03-28:
Just to close this out - I formatted and reimaged the sdcard with the v4 image. Didn't boot of course, so I grabbed the soysauce v5 files and copied them over and holy moly it worked!!!! I did this before when I was trying all the screens (the tool didn't detect my screen so I tried them all) and it did not work...so if anyone is having issues - just reimage the card and put the soysauce dtb files on there again. And yes the joycons are now working! 🍻 
HINT: I took the battery out and saw the model Y3506_V03_20241210 which confirmed it was a Soysauce model.

danielregis81, 2025-07-12:
> Just to close this out - I formatted and reimaged the sdcard with the v4 image. Didn't boot of course, so I grabbed the soysauce v5 files and copied them over and holy moly it worked!!!! I did this before when I was trying all the screens (the tool didn't detect my screen so I tried them all) and it did not work...so if anyone is having issues - just reimage the card and put the soysauce dtb files on there again. And yes the joycons are now working! 🍻 HINT: I have a see-through case - so I took the battery out and saw the model Y3506_V03_20241210 which confirmed it was a Soysauce model.

Hi gr33, what do you mean with " I grabbed the soysauce v5 files ", sorry but I'm new on this matter, I have the same model as you ( Y3506_V03_20241210) and I'm facing exactly the same issue on my joysticks, I confess that I only notice this issue days after the update as I was playing games using only the D-pad. So, could you help me to understand your solution? I have a friend facing the same issue so you can help two lost souls here ehehe. Thanks in advance.

toninhoaugusto, 2025-07-16:
Hello, guys.

After spent a few hours trying to fix the inverted control sticks of my R36S clone (probably a Soy Sauce device), I want to contribute here, just in case there is more people facing the same problem.

**Pre requisites**: you will need to access the root partition of your MicroSD card. There is various different methods to achieve this. I used a application called **Ext2 Volume Manager** on my Windows 10 machine. This partition is in EXT4 format. If you use another OS, you need to do same research to be able to access the root partition.

**Solution 1** - just replace you **es_input.cfg** file with the one I attached here. It's the original one with the X and Y axis corrected to my device hardware. Location of the file: /etc/emulationstation/

**Solution 2** - replace the **Reset EmulationStation Controls.sh** file with the one I attached here. I modified it to not expect you to press any button to confirm the reset of controls for EmulatioStation. It will simply do the reset without any confirmation. You just need to replace the file, put the MicrSD card back on your device, boot it, and go to Options, Advanced and chose Reset EmulationStation Controls. Wait for the script to finish and then reboot the device. After the boot, EmulationStation will ask you to bind the keys again and will update the es_input.cfg file with the new bindings. Location of the file: /opt/system/Advanced/

**Bonus**: as the screen is too small to the input bindings menus texts, the order for the analog sticks is: UP, DOWN, LEFT, RIGHT.

The files was attached with no extensions due to the restrictiosn of GitHub. Add the proper extensions after download them.

God bless you all.

toninhoaugusto, 2025-07-21:
> "replace the **Reset EmulationStation Controls.sh** file content with the script bellow" could you explain to me where I should replace this file? I don't understand much about this and I'm having this problem, I connected the memory card in my PC but I have no idea which file I should replace or modify..

I think a easier solution is the one I put on the last edit of my post: connect your SD card on you PC, and create a new folder in the EASYROMS partition called 1. Inside this folder, create a resetcontrols.sh file with the content I put on my original post 

After that, place the SD card back to your console, and boot it. Go to Options > File Manager and copy your EASYROMS/1/resetcontrols.sh file to root/opt/system/Advanced. The file manager is a little tricky, but you can figure it out for sure 

Then, when you go to Options > Advanced, there will be a script called resetcontrols. Just press A and wait 30 seconds. Then reset your console and you will be prompted to bind all the buttons once the EmulationStation was loaded.

Good luck!

lesharodin, 2025-07-21:
Hi @toninhoaugusto I tried 2 methods and none of theme work.
I changed es_input.cfg but after I go to test contols its still the same. (I have left up down on right stick, and right up down on left stick)
After I launch Reset EmulationStation Controls.sh and reboot console Im not promted to anything, it just boots

lorepappo99, 2025-08-26:
Hi @danielregis81 where did you copied the original soysauce boot files exactly? I'm trying to solve the same problem on my K36 but i can't figure it out

danielregis81, 2025-08-26:
> Hi [@danielregis81](https://github.com/danielregis81) where did you copied the original soysauce boot files exactly? I'm trying to solve the same problem on my K36 but i can't figure it out

you can download [here](https://github.com/user-attachments/files/21387100/Y3506.boot.zip)

danielregis81, 2025-08-26:
> > ah ok, [@lorepappo99](https://github.com/lorepappo99) after you burn the img on the sd card, just copy and repleace all files on boot partition using these files on [Y3506 boot.zip](https://github.com/user-attachments/files/21387100/Y3506.boot.zip) only the SELECT and FN buttons are swapped but the other controls will work
> 
> It seems it doesn’t work for me, my K36 is now stuck on the boot image when I turn it on. What arkos version do you have?

ArkOS 2.0 (06302024-1)(AeUX) on a R36s "soy sauce"

seralbad, 2025-08-30:
fixed 
copying the original boot files to the updated microSD boot

[image]

seltix5, 2025-10-29:
> I think a easier solution is the one I put on the last edit of my post: connect your SD card on you PC, and create a new folder in the EASYROMS partition called 1. Inside this folder, create a [...]

thanks for all your info! that really helped! (the bonus too!)
My board is also: Y3506_V03_20241210 (under the battery)

Based on your instructions I adapted my own wait to do it: I just deleted **/etc/emulationstation/es_input.cfg** using the file manager and restarted the console. at this point because the file is missing ArkOS will prompt the controller setup.