AeolusUX/ArkOS-R3XS #96: I would like to make a PR regarding EmulationStation.

bulzipke, 2025-02-07:
Hello, I have an R36S and came here because I want to use ArkOS.

I made a PR to the EmulationStation source of ArkOS (https://github.com/christianhaitian/EmulationStation-fcamod) and built it on the master branch, but the screen resolution was displayed incorrectly on the R36S.
[image]

I found the EmulationStation branch that you are managing (https://github.com/AeolusUX/EmulationStation-fcamod).
Would it be better to make a PR there? Or is it okay to submit the PR to ChristianHaitian's branch as it is?

I would love to see the following features available on ArkOS and the R36S as well:
https://github.com/knulli-cfw/batocera-emulationstation/pull/25
https://github.com/knulli-cfw/batocera-emulationstation/pull/26
https://github.com/knulli-cfw/batocera-emulationstation/pull/28

Best regards,

AeolusUX, 2025-02-07:
Hi! I believe a PR on christian's repo is better so he can also use it on other devices. I usually just take assets from his builds anyway.

bulzipke, 2025-02-08:
I appreciate your efforts!

closed 2025-02-08 by bulzipke