AeolusUX/ArkOS-R3XS #98: R36S can't work with two memory cards at the same time after online update to 02022025

LobAnovich, 2025-02-11:
After online update of my original R36S with two branded memory cards to version 02022025, it stopped working with two memory cards at the same time? If I select the main card, it shows games only from the main memory card. If I enable support for the first & second memory card, it shows games only from the second memory card. If I enable simultaneous work from two cards, shows games only from the first memory card.
And yes, before the update, I followed the warning and switched to the main memory card

AeolusUX, 2025-02-12:
Have you tried restarting your device after switching to the second memory card?

AeolusUX, 2025-02-12:
> > Have you tried restarting your device after switching to the second memory card? or after you set it to read both cards?
> 
> Yes, of course, both restarting the arkos and completely shutting it down. I even took out the memory cards and tried to run it without them. I have owned this device for a long time and know the main features of its correct use
> 
> Is it possible that this happened because I followed the warning and switched to the primary memory card before installing the update?

No it's not. I just tried this before commenting here and it's perfectly working on 3 of my devices with different installations.

LobAnovich, 2025-02-12:
At the moment, working with two memory cards is completely identical to the stock firmware, except for the presence of the option "simultaneous work with two memory cards". It remains to be hoped that the next update will be able to eliminate the problem that has arisen

AeolusUX, 2025-02-12:
Unfortunately if I can't reproduce it it won't be possible to fix it on my end. Have you tried doing a fresh install altogether?

LobAnovich, 2025-02-12:
I can completely reinstall the system, but I have no desire or time to do this. Since I have only found a few mentions of this problem on the Internet, I am sure that a fresh installation can fix this problem. If there is no way to check or fix the cause, then I will try to find some free time later and perform a clean installation.

AeolusUX, 2025-02-12:
Fair, well since I can't do anything about it without me being able to replicate this I'm closing this. If it still persists after the clean install do let me know.

closed 2025-02-12 by AeolusUX

RyanRoe77, 2025-02-12:
This issue still persists for me. I've done what the above commenter didn't and tried this on a fresh install with 2 different sd cards and I still get the same error. When I revert back to 12242024 build it works fine so I know its an issue confined to this latest update. I'm happy to test if you have a fix.

AeolusUX, 2025-02-12:
Okay I found what's wrong with it thanks for the description it made me realize what was happening. A fix will be added via OTA Update.

LobAnovich, 2025-02-12:
> Okay I found what's wrong with it thanks for the description it made me realize what was happening. A fix will be added via OTA Update.

Great, we are waiting for information about the release of the update. It is very strange that there are almost no reports of the problem from other users. Apparently, few people use two cards with content at the same time

LobAnovich, 2025-02-13:
After updating to version 02092025 the problem is solved

AeolusUX, 2025-02-14:
> After updating to version 02092025 the problem is solved
> 
> Off topic question, but too simple, so I'll leave it here. For my O30S, will I need to select the R36H device type in the settings so that the missing FN button is not used, or is R36H different from O30S?

Should be the same