AeolusUX/ArkOS-R3XS #103: ArkOS R3XS V2.0 (04022024)  ---- Could you please provide the image link

johy-525, 2025-02-19:
Could you please provide the image link？

The following version，
ArkOS R3XS V2.0 (04022024)

Thanks！

AeolusUX, 2025-02-19:
Check links for MEGA from [this.](https://github.com/AeolusUX/ArkOS-R3XS/releases/tag/ArkOS.V2.0.04242024)

closed 2025-02-19 by AeolusUX

johy-525, 2025-02-19:
The version of the link above is not 04022024

AeolusUX, 2025-02-19:
> The version of the link above is not 04022024
I said check MEGA did you check MEGA? I wouldn't link it if it doesn't have it. If you can't help yourself I don't think I can.

johy-525, 2025-02-19:
Okay, I have found it. Thank you very much!

Dattebayooooo, 2025-02-19:
> Could you please provide the image link？
> 
> The following version，
> ArkOS R3XS V2.0 (04022024)
> 
> 
> Thanks！

Why old version?

AeolusUX, 2025-02-20:
> I want to compare whether the A and B button functions are reversed under PSP games. At present, in version 20240929, A is the return function and B is the confirmation function, but in fact, their functions are reversed

You could just easily remap those in the ppsspp menu

johy-525, 2025-02-20:
Is the following directory fixed? ppsspp\PSP\SYSTEM，
Could you please specify which parameter to change? thank you