AeolusUX/ArkOS-R3XS #106: Unable to get ArkOS running

gbuffry85, 2025-03-02:
Apologies if this isn't the place to put this but I am having no luck with getting the Community ArkOS to run on my R36s, which was received a few days ago and I am pretty desperate. Really at my wits end. I am not a complete novice - I have customised the SNES mini with ROMS and emulators so I have a basic understanding, although it is very much basic.

I have detailed a bit more below but the summary is that when trying to boot the device I am generally just getting a black screen with faint white vertical lines (almost barcode like) that change, as if the screen is not displaying what it should be properly. On some occasions I have got an orange coloured screen with vertical lines too. I have waited to see if the device will reboot and the software will load properly but it doesn't. Picture links to imgbb are provided below if they help.

DETAILS

I am on a Mac and my R36s was purchased from here...

https://nl.aliexpress.com/item/1005007226765269.html?spm=a2g0o.productlist.main.7.7f2e6cb35uj5rZ&aem_p4p_detail=20250302005958153257802508560016342699&algo_pvid=ee697c55-12a4-442e-95e8-2b5b1e188ec0&algo_exp_id=ee697c55-12a4-442e-95e8-2b5b1e188ec0-3&pdp_ext_f=%7B%22order%22%3A%2224731%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21EUR%21151.73%2140.59%21%21%211122.26%21300.22%21%40%2112000040440720430%21sea%21NL%210%21ABX&curPageLogUid=bGttDCJNnZ4u&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=20250302005958153257802508560016342699_3

I purchased a brand new SanDISK Ultra 32GB microSD card (Amazon) and this did not work, so I purchased a Samsung Pro Endurance 32GB (Amazon). Neither work.

I have tried using ApplePiBaker as instructed to image one of the builds onto the device. The first time I got the BOOT and EASYROMS disc but nothing happened on startup. Ever since then I am only getting the BOOT disc with ApplePiBaker. When I go to DiskUtility the second disk is just listed as a generic named disc and I cannot mount it. ApplePiBaker expands the initial IMG to fill the 32GB microSD card as it's flashing.

I know that it says there are issues with BalenaEtcher but I thought I would try this to see if there was any change. This did not automatically expand the IMG file but when it mounts it shows the BOOT and EASYROMS each time. When I try to start the device it is still the same.

I have used a number of builds from yourself as well as the ChristianHaitian community build but so far nothing has worked. Similarly I have tried replaced the DTB files with those fprovided and tech4bot but I can't find a working solution.

The microSD card that is in the device is working fine and there is no issues with the screen itself as (most of) the games play perfectly. There seems to be no issues with the device itself that I can see as it's all working fine BUT I am obviously concerned about the longevity of the stock microSD card and also want to modify it to add further games.

I had one build where an intro image appeared but then the screen went to a flashing dash in the top left corner but did not do anything else. Unfortunately I did not record which combination this way.

I am a little confused as to which build I should be using for the device but as I say I have tried a number of different combinations. I suppose my main questions are 1) HELP - how do I get this working, 2) is there a way of knowing which combination of IMG / DTB etc I need for this specific device, 3) is this a device issue or ArkOS not populating properly through ApplePi / balena.

Lastly - in a worst case scenario, can I use ApplePi / balena to clone the stock microSD card to a legitimate microSD card and use it that way? And if I cloned it to (for instance) a 128GB can I then modify the ROMS on it - add new ones etc, even if I can't add new systems?

If there is any further information I can add then please let me know - particularly if there is a way of showing what build the device itself is. I am at my wits end trying to work it out so any help would be GREATLY appreciated and I would be eternally grateful. Very happy to donate for any advice to get it working!

IMAGES

https://ibb.co/Kxwn7TJr
https://ibb.co/prkYNDZh
https://ibb.co/JFMv0q9d
https://ibb.co/N2JNHv1w

With stock SD card

https://ibb.co/yBsMM5c5
https://ibb.co/zTQNTcbd
https://ibb.co/7dnQnRxs
https://ibb.co/fVcnVKpZ

Tom2112, 2025-03-02:
The problem is that the R36S can come with different display panels. If the OS image you boot from doesn't have the correct settings for your specific panel, you'll get a black screen - sometimes with weird lines, sometimes just completely black. Check out the wiki and help threads in the right sidebar of the r/R36S subreddit on reddit. There's a ton of info there and guides to help you get setup.

It's very likely that all you'll need to do is copy the correct .dtb file to your ArkOS cd card after writing the image to it and before you boot it up in your R36S. The trick is finding which panel you have so you can get the right file. There aren't that many versions of panel/files (like 5 or so), so worst case scenario: try each one. See the links below for more info - especially that last link under the "Black Screen" heading and the "Solution ArkOS" subheading.

https://www.reddit.com/r/R36S/
https://handhelds.miraheze.org/wiki/R36S_Handheld_Wiki
https://handhelds.miraheze.org/wiki/R36S_Custom_Firmware
https://handhelds.miraheze.org/wiki/R36S_Guides_%E2%80%90_Tutorials_%E2%80%90_How%E2%80%90To%27s
https://handhelds.miraheze.org/wiki/R36S_Problems_and_Troubleshooting#Black_Screen

gbuffry85, 2025-03-02:
Thank you so much Tom2112 that's really explained it and was super helpful. I've FINALLY managed to get it working with Pack 4, although I have still have some vertical lines down the screen as shown here

https://ibb.co/Lbgv2gW

At least I'm hopefully one step closer!!

Tom2112, 2025-03-03:
Glad to help. Did you copy the dtb file from the original sd card, or did you download panel 4 drb file? If you downloaded it, then you might want to try the original one from your original sd card, especially if when booting with the original you don't get those lines. There may be something subltely different about your panel from the typical panel 4. It's worth a shot.

closed 2025-03-23 by AeolusUX