AeolusUX/ArkOS-R3XS #107: Missing Python 'inputs' package in ArkOS AeUX 02092025 prevents /usr/local/bin/auto_suspend.py from running

cowbutt, 2025-03-02:
I'm not sure if this is a deliberate kludge to easily disable the Auto Suspend feature, or if the presence of the Python inputs package ( https://pypi.org/project/inputs/ ) has unwanted side effects:

$ /usr/local/bin/auto_suspend.py
Traceback (most recent call last):
  File "/usr/local/bin/auto_suspend.py", line 3, in <module>
    from inputs import get_gamepad
ModuleNotFoundError: No module named 'inputs'

Dattebayooooo, 2025-03-03:
Can you reinstall it via pip?

cowbutt, 2025-03-03:
> Can you reinstall it via pip?

The ArkOS AeUX 02092025 image does not include the inputs package, or pip.

AeolusUX, 2025-03-04:
huh that's weird, it was installed [here](https://github.com/AeolusUX/ArkOS-R3XS-Updater/blob/43e8a5eb53f4655d887a603ab6404e02f02490bd/Update-R3XS.sh#L692). I'll check my devices maybe the image I use as base failed to install it.

AeolusUX, 2025-04-12:
I added code to reinstall it if not found on the system for todays release.

closed 2025-04-12 by AeolusUX