AeolusUX/ArkOS-R3XS #110: New version of R36S (Panel) and original DTB files for it.

tomekfm, 2025-03-12:
Hello!
I bought 2 pcs R36S from Factory Game Direct Store [LINK](https://aliexpress.com/item/1005007829085568.html) and they came with a new green LED and new display type. New LCD is less contrast and saturated:
[image]
[image]

[DTB Identify Tool](https://aeolusux.github.io/ArkOS-R3XS/tools/dtbIdentify.htm) doesn't recognize it yet and returns: **The uploaded data does not match the content of any pre-existing file.** 
When I used the original DTB files, ArkOS worked fine (files in attachment), so I decided to share them; maybe they can help someone.

File: [DTB.zip](https://github.com/user-attachments/files/19216516/DTB.zip) contain:
- rk3326-rg351mp-linux.dtb
- rg351mp-kernel.dtb
- gameconsole-r36s.dtb
- boot.ini

Roll2Me, 2025-03-12:
I have the same files you do also! I just got mine last week and when I used tool I also get " The uploaded data does not match the content of any pre-existing file."  I wanted to make a new microSD card with the OS after reading how they used cheap card and it fails a lot! But since I don't know what to use now I'm stuck!

tomekfm, 2025-03-13:
I finally copy all of them to the BOOT partition main folder and overwrite all existing files, but I tested with use only rk3326-rg351mp-linux.dtb and ArkOS installed successfully too, so it can be enough.

AeolusUX, 2025-03-13:
Would you mind sharing the whole boot partition aside from the DTB files?

tomekfm, 2025-03-13:
I had 2pcs, so i did image of both of them. Boot partitions should be same, but checksum is little differ.
[https://www.mediafire.com/file/58tx6n3pm78rm6u/BOOT.zip/file](https://www.mediafire.com/file/58tx6n3pm78rm6u/BOOT.zip/file)

21milon, 2025-03-16:
@tomekfm 

I had the same issue and indeed this solved it. 
1. Fresh V4 install
2. Copied the 4 files from my own original backup (your files didnt work).

Thanks for the idea and solution!

Jeavo, 2025-03-18:
@21milon I have the same issue. Bought from BOYHOM last week so don't think it's a clone.

So I just need to perform a regular OS flash onto my new card?
And which 4 files are you referring to please?

Thanks

gr33k, 2025-03-18:
I received one a few days ago and there is a text file indicating V5 (R36S V5.txt). 
None of the new or old screen files worked beyond a blank screen. Using my originals - it booted up. Here are the files:

[BOOT.zip](https://github.com/user-attachments/files/19326907/BOOT.zip)

21milon, 2025-03-18:
@Jeavo Yes, just do a backup of your original card or better use a different one, make a OS flash V4 flash and copy 4 files from your original card/backup. Files are those which tomekfm mentioned:

rk3326-rg351mp-linux.dtb
rg351mp-kernel.dtb
gameconsole-r36s.dtb
boot.ini

Only thing which upset me is, that key shortcuts doesnt work (exit game etc.)

21milon, 2025-03-18:
@gr33k are your shortcuts working?

gr33k, 2025-03-18:
@21milon - I'm new to this thing, but seems like the function button is working. Still transferring roms to SDcard2 and will be able to test better once completed. The light is on Red though despite trying to switch it to blue (which menu has 'change led to red' currently despite already being red lol). I ran it twice (switch to red and switch to blue, but no changes happen).
There may be other oddities yet to be discovered. I'll play with it a bit more and report back.

Jeavo, 2025-03-18:
I just installed Panel 4 (V5) and booted up to check if it worked before I replaced your suggested files.
It seems to be working perfectly fine - will update if I see any issues though.

gr33k, 2025-03-19:
@21milon it almost seems like select is the hotkey and FN is the select button from what I'm experiencing thus far (like NES for example).
Some tool menus (like the reset global hotkeys advanced option) says press A to continue or B to cancel - no button works here.
I can't reset emulationstation controls either.
Joycons are not pointing in correct direction
Will keep testing...but wish there was an easy way to remap the controller buttons like in a RasPi emulationstation menu..

marccarbonell12, 2025-03-20:
Hi there, new to this thing too but i did get a new r36h last week, came with green led light and im getting same message when usin screen detecting tool. 

I have a new 128 sd card but so far i just been able to flash the original copy of the image that came with the original sd card. 
Problem im having is the EASYROM partition not expanding at all after first reboot and remains as if i had the 64gb card still. 

Any clue on how to fix that? Do you recommend trying a fresh install with the panel 4 version from arkos?

Tom2112, 2025-03-20:
> Hi there, new to this thing too but i did get a new r36h last week, came with green led light and im getting same message when usin screen detecting tool.
>
> I have a new 128 sd card but so far i just [...]

If you've got a spare SD card, I would definitely try a panel 4 version. Other than a little time, you're not out anything. The last seven R36H units I've received were all green LED models and work great with panel 4 version of the OS. I didn't even bother to check their dtb file to see which version panel they have, figuring just try my existing panel 4 version SD card and it worked. But I'd bet yours will too.

marccarbonell12, 2025-03-22:
Thanks mate, I ll try installing the arkos for panel 4 on a new SD then see what happens... hopefully i dont get the black screen problem

AeolusUX, 2025-03-23:
This has been already added to the tool days ago. Thanks for providing these files.

closed 2025-03-23 by AeolusUX

kostasmm, 2025-06-15:
@tomekfm 

Thank you so much! Your solution worked perfectly for me. 

I also have the new R36S with green LED from Shi Baoju digital store, and I was getting the same "The uploaded data does not match the content of any pre-existing file" error from the DTB identify tool. I was starting to worry that I had received a clone device.

Following your advice, I backed up my original 4 files:
- rk3326-rg351mp-linux.dtb
- rg351mp-kernel.dtb
- gameconsole-r36s.dtb
- boot.ini

Then I flashed ArkOS Panel 4 (V5) and copied these 4 files to the BOOT partition, overwriting the existing ones. The device booted up perfectly!

This saved me a lot of time and frustration. Thanks again for sharing your findings with the community!

AeolusUX, 2025-06-15:
> [@AeolusUX](https://github.com/AeolusUX)
> 
> Hi! I just got my R36S with green LED from Shi Baoju digital store and I'm still getting "The uploaded data does not match the content of any pre-existing file" from the DTB identify tool at https://aeolusux.github.io/ArkOS-R3XS/tools/dtbIdentify.htm
> 
> Maybe mine is a newer variant? I successfully installed ArkOS Panel 4 (V5) using my original DTB files, so I'd like to share them to help update the tool for others who might have the same issue.
> 
> Here are my original 4 files in a zip: https://drive.google.com/file/d/1s1droNMAAx8xgigcBmPNT4XbppBv3AXR/view?usp=sharing
> 
> * rk3326-rg351mp-linux.dtb
> * rg351mp-kernel.dtb
> * gameconsole-r36s.dtb
> * boot.ini
> 
> My device details:
> 
> * Purchased: 05/June/2025 from Shi Baoju digital store
> * Green LED model
> * Works with Panel 4 (V5) + these original DTB files
> 
> Hope this helps improve the tool's detection! Let me know if you need any other files or information from my device.
> 
> > This has been already added to the tool days ago. Thanks for providing these files.

Did you upload the gameconsole-r36s.dtb file instead of the rk3326-rg351mp-linux.dtb file?

rodinaelghonimy2011-jpg, 2025-09-27:
> I have the same files you do also! I just got mine last week and when I used tool I also get " The uploaded data does not match the content of any pre-existing file." I wanted to make a new microSD card with the OS after reading how they used cheap card and it fails a lot! But since I don't know what to use now I'm stuck! I don't think it's a clone because I got it from BOYHOM Store which is a recommended store.
> 
> Please let me know which one you can use to flash new firmware to setup new OS microSD card!

Look if yours has a qc passed sticker it is likely a clone I bought mine from BOYHOM store too and found the hard why it is a clone

rodinaelghonimy2011-jpg, 2025-09-27:
Look if yours has a qc passed sticker it is likely a clone I bought mine from BOYHOM store too and found the hard why it is a clone