AeolusUX/ArkOS-R3XS #112: Can't set new favourites

djcruzzab, 2025-03-14:
With the new version, all favorites I set are erased when the system is rebooted. Default favorites are still there. Apparently, I am not the only one suffering this (https://www.reddit.com/r/R36S/comments/1fe2z9u/favourites_emptying_itself_when_turned_off/)

cowbutt, 2025-03-14:
Is the SD card where your ROMs are stored still writeable? Or has it failed read-only?

djcruzzab, 2025-03-14:
Yes, I can generate savefiles without any problem, so the SD is writeable

AeolusUX, 2025-03-23:
According to your linked reddit post this has been present since 6 months ago then that means it's nothing to do with the new version. If it only happened to you with this new version you might need to check your sd card if it's really legit. There are tons of fake cards out there. I've tested atleast 3 devices and I could not replicate this same issue.

AeolusUX, 2025-04-14:
This has been fixed by the new update.

closed 2025-04-14 by AeolusUX

djcruzzab, 2025-04-16:
I have just tried the new update and I still have the same issue