AeolusUX/ArkOS-R3XS #113: Rumbling support R36H

ARPABoy, 2025-03-17:
Hello,

I have loaded ArkOS in my R36H successfully and the whole system works correctly. I have soldered a vibrator in the MOTO motherboard location but it's not controlled correctly, it starts rumbling from the power on moment without stopping anymore.

I tried to locate the GPIO that controls this but unsuccessfuly. Any idea or suggestion?
Maybe the DTB file should be altered in any way?
Maybe some u-boot configuration changes?

AeolusUX, 2025-03-23:
No such thing, adding a rumble just works when you add them no need to configure anything. If you have any issues it's either your vibration motor or you could check the polarity and the quality of your solder. There had been reports of vibration motors that does exactly this and when replaced with another vibration motor it works flawlessly. So I suggest trying a different motor or double check the soldering. This is not in any way affected by the software.

closed 2025-03-23 by AeolusUX

ARPABoy, 2025-03-23:
> No such thing, adding a rumble motor just works when you add them no need

R36HRumble.mp4
<https://drive.google.com/file/d/1wxSSFujFcNmmL3ir_PS3bQFk6sLoSVWU/view?usp=drive_web>
 R36SRumble.mp4
<https://drive.google.com/file/d/1aiSd9bgjVvzrcnNuHy4egHqXitNJ5hLq/view?usp=drive_web>
I have soldered wires instead of the motor in order to read voltage using a
multimeter.

When I power on the console there's 3.6 volts approximately. So another
motor wouldn't solve the problem as voltage is there.

I have soldered the same kind of motor in R36S consoles without issues, but
R36H is different in some way.

I attach both voltage videos of R36S and R36H.

I think it's something DTB files related problem.