AeolusUX/ArkOS-R3XS #118: Fix for cheap WiFi dongles

laurorual, 2025-04-02:
I use one of those really cheap Aliexpress WiFi dongles that mounts a USB storage before actually function as a WiFi dongle (for installing drivers on Windows) and, by default, it doesn't work on ArkOS.
I found a fix for this on this Reddit post: https://www.reddit.com/r/SBCGaming/comments/17o6nna/arkos_and_cheap_wifi_dongles_working/
And apparently Rocknix also has this fix implemented by default.
Could you implement this on ArkOS as well? It would be cool having this working by default everytime i have to reinstall the OS into a new SD!

AeolusUX, 2025-04-11:
Interesting stuff but I have no way to test this as I don't have those dongles. This post might help with other stuff though so thanks for bringing this to my attention.

closed 2025-04-11 by AeolusUX

laurorual, 2025-04-16:
No problem! 
In the future, if you want someone to beta test this implementation, I can do it for you.

Dattebayooooo, 2025-04-16:
I can as well test, I have unsupported wifi dongle as well.

laurorual, 2025-04-23:
> I can as well test, I have unsupported wifi dongle as well.

There's a way to create a script to fix this issue, tho i haven't tested it myself since i usually do the fixes by myself.
But if you haven't done the fix yet, you can try to run the following script on your device:

`#!/bin/bash

# Create the USB modeswitch configuration file
echo "# AC600 Wifi Dongle
TargetVendor=0x0bda
TargetProduct=0x1a2b
StandardEject=1" | sudo tee /usr/share/usb_modeswitch/0bda:1a2b > /dev/null

# Check if the file was created successfully
if [ -f "/usr/share/usb_modeswitch/0bda:1a2b" ]; then
    echo "USB modeswitch configuration file created successfully."
else
    echo "Failed to create USB modeswitch configuration file."
    exit 1
fi

# Modify the udev rules file
UDEV_RULES_FILE="/lib/udev/rules.d/40-usb_modeswitch.rules"
TEMP_FILE=$(mktemp)

# Check if the file exists
if [ ! -f "$UDEV_RULES_FILE" ]; then
    echo "Error: $UDEV_RULES_FILE not found."
    exit 1
fi

# Find the line with LABEL="modeswitch_rules_end" and insert our rule before it
awk '/LABEL="modeswitch_rules_end"/ {print "# Realtek 8811cu wifi usb"; print "ATTR{idVendor}==\"0bda\", ATTR{idProduct}==\"1a2b\", RUN+=\"usb_modeswitch \047/%k\047\""; print}1' "$UDEV_RULES_FILE" > "$TEMP_FILE"

# Replace the original file with our modified version
sudo cp "$TEMP_FILE" "$UDEV_RULES_FILE"
rm "$TEMP_FILE"

echo "Udev rules updated successfully."
echo "You may need to restart your system or reload udev rules for changes to take effect."`

Just copy all this text into notepad, save it as setup_usb_modeswitch.sh or whatever, put it inside tools folder and run it on your console, see if it helps!