AeolusUX/ArkOS-R3XS #120: Sleep Mode & Safe Shutdoen now working anymore?

QUIZASERAQ, 2025-04-12:
FN and Power = Safely shutdown device
Power (Short Press) = Put device to sleep

--> these function do not work anymore since I updated to v2.0 (03/29/2025) (via Options-> Update).
--> Do I need to change anything in OPTIONS or MAIN MENU to reactivate power btn short press functions?

Thx!

AeolusUX, 2025-04-12:
This is a known bug for now. I'll roll out a fix for it in a few.

Plaque-fcc, 2025-04-14:
My experience is, if you do a fresh install, the bug appears.
If you imply upgrade from 02092025 to 03302025, then the bug doesn't appear.

QUIZASERAQ, 2025-04-14:
Well.. I UPDATED.. 02092025 -> 03292025 -> 03302025 !!

AeolusUX, 2025-04-14:
Builds and updater is now updated.
Unfortunately it's become impossible to remove this bug for OTA update users so there's no file I can rollback to let you fix it. 
You guys have to reflash your sd cards for the bug to go away.

closed 2025-04-14 by AeolusUX

xes, 2025-04-15:
Sorry, which build contains the fix?

mmdudek, 2025-04-17:
Do we simply re-download the 3302025 build and flash it to SD? I re-downloaded the 3302025 build today and the byte count is the same as the 3302025 build I downloaded last week, which makes me think I'm doing something wrong...

Weigh13, 2025-04-22:
> Do we simply re-download the 3302025 build and flash it to SD? I re-downloaded the 3302025 build today and the byte count is the same as the 3302025 build I downloaded last week, which makes me think I'm doing something wrong...

You're not the only one wondering. It still seems like the update wasn't changed. The issue still persists. Really wanting to get back this feature!

mmdudek, 2025-04-22:
I did re-download the 3302025 build and re-flashed it to my sd card and voila - the power button now works like it should! Fn+Power for safe shutdown, short press Power for sleep mode.

Give it a shot.

Weigh13, 2025-04-22:
Awesome, thank you for clarifying!