AeolusUX/ArkOS-R3XS #121: EmulationStation in 03302025 uses much more memory than in 02092025

cowbutt, 2025-04-14:
I've just updated from 02092025 to 03302025. When I switch back to using SD2 for ROMs, EmulationStation startup loops after Movies->Favorites->Starting UI. As I have in excess of 300K ROMs on a 256GB card in SD2, I believe this is because it is trying to use more RAM than the 1GByte my R36S has present; monitoring it using top during startup sees its usage rise to over 1.7Gbyte before it restarts.

This is probably my own silly fault for having too many ROMs on a single SD card, but it would be nice if there was a more graceful failure mode, or even if EmulationStation's memory usage could be reduced again. I'm going to restore my 02092025 backup.

AeolusUX, 2025-04-15:
Alternatively you could go for the old emulationstation binary from the 02092025 build.

cowbutt, 2025-05-15:
FYI, I've now removed ~250K 8 bit micro ROMs from my main ROMs SD2 card, as even with 02092025, EmulationStation's memory usage was preventing some PortMaster ports from running without swap.