AeolusUX/ArkOS-R3XS #122: Power Button Won't Sleep Device/Quick Mode Won't Shut Down

staticcornflake, 2025-04-14:
Hi,

Sorry if this is a dumb one, I recently flashed the OS and everything is working great, but my power button (a tap) doesn't put the device to sleep, and I've enabled Quick Mode, can't get it to shutdown with FN+R3 or Power Button+R3.

Are there any files or settings I may have missed on this? The power button boots up the device and will do a hard shutdown. Thanks!

retrorobdog, 2025-04-14:
I actually just joined Github and created an account so I could come here to say the same thing. I can't use Sleep mode at all. I have not enabled quick mode on this device because I previously bricked a device and had to use ArkOS-R3XS to replace the bricked firmware and fix the device. Quick mode does have lots of issues but I dont think its related to this one. Really wish I could use Sleep mode but since updating I can not.

retrorobdog, 2025-04-14:
Also to clarify. I have gone into the advanced settings and toggled Switch power Button Tap to Off. Ive done it with it on and with it off and sleep mode doesnt work on either. And its not powering off the device.

retrorobdog, 2025-04-15:
@AeolusUX Sorry for the tag just wanted to find out if you had seen this and had any information or a possible fix?

AeolusUX, 2025-04-15:
https://github.com/AeolusUX/ArkOS-R3XS/issues/120

closed 2025-04-25 by AeolusUX